jotego/jtcps #33: CPS1 core - Street Fighter 2 Turbo Hyper Fighting: The speed differential: The issue

fluxcorenz, 2020-05-22:
I've been trying to get good timing measurements to ensure the CPS1 core is as accurate timing-wise as possible, and Street Fighter 2 Hyper Fighting has been a thorn in the side of emulation for about 20 years now.

Video comparing 20200501 mister CPS1 core vs arcade board:
https://www.youtube.com/watch?v=7f4TsJQQSB0

From the sync time (0m21s) to the start of the 'continue' countdown screen:
Arcade board: 250s
Mister: 267s

Making a speed differential of roughly 6.8%

ROM revision is same: 921209 ETC

DIP settings (shown at end of video) are not exactly identical (I couldn't set the Mister dips the same), but I did a later test with arcade dips matching the Mister settings and the speed difference between the two arcade captures was around 14frames over the ~1500 frames of test time - probably within the general variation of timing of the arcade board, if I were to make more captures. 

The same test on the earlier 20200410 rbf (against the same arcade footage) showed slightly faster mister speeds (i.e. the core has slowed down between 20200410 and 20200501)
https://www.youtube.com/watch?v=A4euNTIXxXw
From the sync time (0m19s) to the start of the 'continue' countdown screen (different from the above due to the sync point):
Arcade board: 252s
Mister: 266s

Roughly 5.5% slower

Furthermore, enabling the 'turbo' mode in the 20200501 mister cps1 core almost matches the arcade board speed (mister being faster by 15 frames or so):
https://www.youtube.com/watch?v=gs-IZaOrSlc

I'm working on getting footage from other people's arcade boards to gain confidence in general, to prove my arcade board isn't anomalous.

Critiques on testing methodology etc welcome

jotego, 2020-08-07:
I was thinking today that maybe the difference could come from different frame rates. So I verified a couple of things. The conclusion is that the frame rate does not seem to be a factor.

Schematics show that the crystal oscillator is different for the CPU and the GPU (CPS-A chip). The CPU uses a 10MHz crystal in most CPS1 games and allegedly a 12MHz one in some titles. The CPS-A chip uses a 16MHz crystal. The frame rate is set by the CPS-A chip so there is no reason to think that the frame rate would be different.

The frame rate on my SF2WW board -measured through DMA periods- is 59.78Hz. The JTCPS1 core outputs 59.63Hz. The difference is 0.25% and falls within the error tolerance of the equipment used.

[image]

jotego, 2020-12-16:
I have fixed a problem recently in the CPU that should bring back some of the speed. However, the main point is probably the DMA timing. I haven't uploaded the new RBF as it may be troublesome to 128MB module owners.

fluxcorenz, 2020-12-16:
I agree with the earlier point about frame rates - all the CPS1 Street Fighter boards run at the same graphics frame rate - SF2 Turbo uses techniques to run faster internally (https://game.capcom.com/cfn/sfv/column/131667) but still outputs at ~60Hz

I'm happy to give the rewritten memory controller version a test if you supply the RBF - I do use a 128MB module but will keep that in mind regarding any issues that might pop up.

jotego, 2021-01-30:
One year later... I have news.

During this time I have verified that the dash boards only have a faster clock. There are no other differences.

Grego2D and I designed a test software to run on real hardware and measure the time. Without going into much detail, the results can be summarized in:

MAME 9090
JTCPS1 Turbo 7519
REAL PCB 6293
JTCPS1 No turbo 5090

So the core is +20% faster with turbo, and 20% slower with no turbo. I did a first improvement to the DMA, and got it from 20% to 15%. I will be closing the rest of the gap in the coming days.

jotego, 2021-01-31:
I have improved the bus wait logic. The time difference is now 0.08%. RBF already available in jtbin.

fluxcorenz, 2021-01-31:
https://www.youtube.com/watch?v=2nJiMnxXYVA

Original time loss over all the idle rounds was 24 in-game seconds, now down to 3.
Total frame loss is 97 frames over 14999, ~0.65% slower than arcade.

Pretty huge improvement!

jotego, 2021-02-01:
It's a great comparison. Thank you so much for re-doing it. 

Some of the error will be due to the difference between the  PCB aged oscillator and the crispy shiny 8MHz sharp oscillator in MiSTer. 

We are going to take that factor out in a future test.

alessioscand, 2021-02-01:
Hello, 
are these improvements valid also for SIDI? 
Thanks

jotego, 2021-02-01:
Of course, the updated core is in the jtbin repository.

jotego, 2021-03-09:
Speed matched.

closed 2021-03-09 by jotego