jotego/jtcps #34: [CPS1] - Options menu corrupted in MIST unless some dip switch lines removed from MRA

glopgar, 2020-05-22:
When generating some .ARC files for MIST, looks like some `<dip ` switch tags are not correctly processed by the `mra.exe` file. If not removed from the `.mra` files, the core options menu gets corrupted. When removed the core options work OK and the config file is loaded correctly on subsequent loads.

I'm using last jtbin MIST core from Patreon and latest MRA.exe: https://github.com/sebdel/mra-tools-c/commit/7569ae9899226f58a9404068139d0a5cfe9af66c

Lines that need to be removed in order for the core options to work:

**Final Fight (World)**

`<dip bits="0,2"	 name="Coin A" ids="4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/6 Credits,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"/>`
`<dip bits="3,5"	 name="Coin B" ids="4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/6 Credits,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"/>`
`<dip bits="6"	 name="2 to Start, 1 to Cont." ids="On,Off"/>`
`<dip bits="13,14"	 name="Bonus Life" ids="None,100k and every 200k,200k,100k"/>`
        

**Ghouls'n'Ghosts (World)**

`<dip bits="0,2"	 name="Coin A" ids="4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/6 Credits,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"/>`
`<dip bits="3,5"	 name="Coin B" ids="4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/6 Credits,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"/>`
`<dip bits="12,13"	 name="Bonus Life" ids="40K, 70K and every 80K,20K, 50K and every 70K,10K, 30K and every 30K,30K, 60K and every 70K"/>`

jotego, 2020-05-22:
I'm trying to have this fixed in the mra tool so the ARC files come out clean.

jotego, 2020-09-10:
I am being careful about newly generated MRA files but for old ones I am going to simply accept pull-requests in jtbin repository from now on. There have already been a number of PR with old MRA files and I think it's better to maintain them there.
MRA files can be modified by users with little technical knowledge about CPS1 system internals. So it frees me up a bit to simply accept contributions.
Thank you for reporting this and please feel free to submit a correction via PR.

closed 2020-09-10 by jotego