jotego/jtcps #36: Complet mame key mapping

funkycochise, 2020-05-22:
By now only the first 3 buttons for player 1 and 2 are mapped
P1 : B1 (Left Ctrl), B2 (Left Alt), B3   Space
P2 : B1 (A), B2 (S), B3 (Q)

Additional keys needed for kick harness are not recognized 
P1 : B4 (Left Shift), B5 (Z) , B6 (X)
P2 : B4 (W) , B5 (I) , B6 (K)

jotego, 2021-02-02:
Sorry for the long wait.

MAME default keys are supported now:

6 buttons for 1P
5 buttons for 2P
3 buttons for 3P

The service key (9) and test key (F2) have been added too.

This will come automatically in future updates.

closed 2021-02-02 by jotego

prrole, 2021-02-02:
Player 2 button 6 would make a lot of sense to add as K, even if it's not default in MAME - it is the de-facto standard.

Reference:

I-PAC, see, see I-PAC installation tab:
https://www.ultimarc.com/control-interfaces/i-pacs/i-pac2/

J-PAC, see J-PAC Installation tab:
ultimarc.com/control-interfaces/j-pac-en/j-pac-jamma-interface

JammaSD
arcadexpress.com/documentos/JammASD.pdf

JVS-PAC:
jvspac.kirurg.org/?page=operation

JVS-PAC 2:
irkenlabs.com/jvs-pac2/keyboard-operation

el-goe, 2021-02-02:
It would be nice to also be able to remap them on joysticks.

jotego, 2021-02-03:
I have added K (to the source code, not RBF files yet). I think it won't be a problem to share it between 3P and 2P because I don't know of 6-button games for three joysticks.

I cannot really add local mapping in the core without much havoc at all levels (user, developer and MiSTer framework). The current approach is this one:

1. MiSTer provides PS2 keyboard and gamepad inputs to the core
2. MiSTer can map keypresses to gamepad inputs transparently to the core
3. The core can map keypresses to gamepad inputs too

Note that 2 and 3 can potentially collide. An option to disable 3 may be advisable, but I don't want to pollute the OSD with more options and so far no one has complained about a clash of key mappings. Maybe they will now that F2 is assigned to the test key.

F2 as a test key is useful in the CPS1.5 and CPS2 cores and had been requested.