jotego/jtcps #37: CPS1 Investigate MRA support for rare Capcom test build 'The World Warrior & Champion Edition'

mholzinger, 2020-05-24:
To investigate - Support for for rare Capcom test build 'The World Warrior & Champion Edition'

Source: Arcade projects user posts a new Street Fighter arcade build has been uncovered.

Post to the project and the forum are here: https://www.arcade-projects.com/forums/index.php?thread/12828-rare-game-the-world-warrior-champion-edition/

```Street Fighter II - The World Warrior &Champion Edition

Street Fighter 2 910522 - Etc. build

A-board 89626A-4 DASH
B-board 91634B-2
C-board PPU B-11
```

- According to the forum post, this build may only work patched in ram and was a test build by capcom.

- Dump of files attached to this issue post.
[sf2-05222020.zip](https://github.com/jotego/jtbin/files/4670644/sf2-05222020.zip)

[image]

- Also to note from Post- This may appear on a cloned Pang 3 board.

jotego, 2020-09-10:
Thank you. I have added it to jtbin. Please make PR there if you have files or changes to add.

closed 2020-09-10 by jotego

cracklejoy, 2020-09-25:
On first glance this appears to be one of yumeji's boss hacks and there are several in hbmame.  I have not checked byte for byte but I highly doubt that this is a test build without further information. I just finished a playthrough. That url leads nowhere.