jotego/jtcps #38: Star field in Strider

ArtemioUrbina, 2020-05-30:
The star field doesn't twinkle in MiSTer and some stars are rendered as lines.

Video comparing both: https://www.youtube.com/watch?v=p-YIV0Z6YGY

Thanks for the great work!

jotego, 2020-06-08:
Thank you so much for the video.
I don't have the exact algorithm for the stars. Nobody does as of today. But I would have expected them to twinkle in the same way. That's something I want to look at.

ArtemioUrbina, 2020-06-08:
Let me know if you want a direct capture, I can use an OSSC/XRGB/XPC-4 + X-Capture 1

wnka, 2021-02-11:
Issue impacts Forgotten Worlds star fields too

jotego, 2021-02-11:
It does, yes. I was expecting the CPS-B chip decapping to show the logic for this but the circuits unveiled do not seem to be the full story. Part of it is in the other CPS-A chip, which is not known yet. 
I will check again the CPS-B schematics and see if I can apply something here.

jotego, 2021-09-29:
The stars are now documented thanks to Loïc Petit. There are a couple of text files [here](https://gitlab.com/loic.petit/cps2-reverse/-/blob/master/DLs/DL-0311/doc/stars.md) and [here](https://gitlab.com/loic.petit/cps2-reverse/-/blob/master/DLs/DL-0921/doc/star-raster.md) but also all the schematics for CPS-A and CPS-B chips.
I'm working on them now and will update the core this week.

ArtemioUrbina, 2021-09-29:
Awesome!

jotego, 2021-10-01:
This is fixed now. I have to make sure the changes didn't affect CPS2 so I will upload the source code later.
I'm not sure about Forgotten Worlds now. The circuit is correct, but the core may be reading the ROM at the wrong place. I need some original pictures for reference.

Strider is perfect. Arcade:
[image]

Core:
[image]

closed 2021-11-10 by jotego