jotego/jtcps #40: CPS1 Carrier Air Wing sprite priority bug

PhantombrainM, 2020-06-08:
Yesterday I played Carrier Air Wing all 9 (or 10?) rounds with Continue dip on.

I noticed when the plane is fueled up (all three rounds or so?) that the pipes are flying through the air and I thought it looks weird.

So I opened Youtube and found the same behaviour but also some YT Vidoes without the problem. I think it's a bug.

Wrong (like on MISTer):
https://www.youtube.com/watch?v=ev1rgJm6uWw&t=545s

Seems right:
https://www.youtube.com/watch?v=T2ffq0MHzGI&t=306s

Also I lost control during the game at stage 7 or 8 (no control anymore and I died). Then I noticed I need to press start again and from that on I was player two (?) until the end of the game. Strange. I don't know why this happened and if it was a one time issue or not. Maybe I'll play it again tonight.

jotego, 2020-06-08:
This was originally issue #28 but it is described better here. There is a pic here:
https://clips.twitch.tv/IntelligentPiercingCoyoteFloof
The problem is that when I looked at the internal video signals, the drawing actually makes sense as it is. So I think I may be missing a piece of the puzzle. MAME gets this right but gets the NEMO letters wrong, which are very obvious too.
This will get sorted when I finally receive the EPROMs I've been waiting for.

The second issue you mentioned sounds more like a software problem. Maybe a different version of the game will work ok.

c0d3h4x0r, 2020-06-13:
I just noticed the same bug and took a photo of it.

[image]

jotego, 2020-12-08:
Technical notes:
Layer use:
SCR1 = text
SCR2 = big plane
SCR3 = clouds
OBJ = small plane and hoses
Control registers:
PRIO0 = 0000
PRIO1 = F7FF
PRIO2 = 03FE
PRIO3 = FFFF
Layer control = 1A5A

PhantombrainM, 2021-01-31:
FYI. Tested. CPS 01/31/2021 Carrier Air Wing sprite priority glitch after 2rd Level is still present. But new DMA runs butter smooth now.

asturur, 2021-07-28:
Also the capcom arcade museum for consoles do this right. in the 30 June version of the core the bug is still present.

jotego, 2021-12-17:
Fixed in 0fdcb02. But NEMO letters during the demo now look like in MAME. Maybe there are really supposed to be like that?

asturur, 2021-12-17:
Oh wow is cool that is been fixed. What was the issue? Does it affects more than just carrier air wing and nemo?

jotego, 2022-06-25:
The commit 0fdcb02 that *fixed* this issue, broke 
- "Three wonders" intro. See #131 
- Final Fight and Varth alignment. See #132
The priority mux section was reverted in edd0cb9, reopening this issue.

asturur, 2022-06-25:
gonna have 2 copies of the core 😏