jotego/jtcps #41: Many MRAs are missing MameVersion

c0d3h4x0r, 2020-06-09:
**/media/fat/Scripts/mra_rom_check.sh** says:
```
There are issues with one or more of your MRA or MAME files

Please gently encourage Developers and Maintainers of MRA files to correct files with broken XML, missing CRC values and missing MAME versions values to correct their MRA files
```

**/tmp/mra_rom_check.log** says:
```
/media/fat/_Arcade/1941  Counter Attack -World 900227-.mra:No MameVersion
/media/fat/_Arcade/Adventure Quiz Capcom World 2 -Japan 920611-.mra:No MameVersion
/media/fat/_Arcade/Captain Commando -World 911202-.mra:No MameVersion
/media/fat/_Arcade/Carrier Air Wing -World 901012-.mra:No MameVersion
/media/fat/_Arcade/Daimakaimura -Japan Resale Ver.-.mra:No MameVersion
/media/fat/_Arcade/Daimakaimura -Japan-.mra:No MameVersion
/media/fat/_Arcade/Dynasty Wars -USA, B-Board 89624B- -.mra:No MameVersion
/media/fat/_Arcade/Final Fight -World, set 1-.mra:No MameVersion
/media/fat/_Arcade/Forgotten Worlds -World, newer-.mra:No MameVersion
/media/fat/_Arcade/Ganbare! Marine Kun -Japan 2K0411-.mra:No MameVersion
/media/fat/_Arcade/Ghouls'n Ghosts -World-.mra:No MameVersion
/media/fat/_Arcade/Ghouls'n Ghosts -patched-.mra:No MameVersion
/media/fat/_Arcade/Gulun.Pa_-_Prototype_1993_-_Japan_931220_L-.mra:No MameVersion
/media/fat/_Arcade/Knights of the Round -World 911127-.mra:No MameVersion
/media/fat/_Arcade/Magic Sword  Heroic Fantasy -World 900725-.mra:No MameVersion
/media/fat/_Arcade/Mega Man  The Power Battle -CPS1, USA 951006-.mra:No MameVersion
/media/fat/_Arcade/Mega Twins -World 900619-.mra:No MameVersion
/media/fat/_Arcade/Mercs -World 900302-.mra:No MameVersion
/media/fat/_Arcade/Nemo -World 901130-.mra:No MameVersion
/media/fat/_Arcade/Pang! 3 -Euro 950601-.mra:No MameVersion
/media/fat/_Arcade/Pnickies -Japan 940608-.mra:No MameVersion
/media/fat/_Arcade/Quiz %26 Dragons  Capcom Quiz Game -USA 920701-.mra:No MameVersion
/media/fat/_Arcade/Quiz Tonosama no Yabou 2  Zenkoku-ban -Japan 950123-.mra:No MameVersion
/media/fat/_Arcade/Street Fighter II  The World Warrior -World 910522-.mra:No MameVersion
/media/fat/_Arcade/Street Fighter II'  Champion Edition -World 920513-.mra:No MameVersion
/media/fat/_Arcade/Street Fighter II'  Hyper Fighting -World 921209-.mra:No MameVersion
/media/fat/_Arcade/Street Fighter Zero -CPS Changer, Japan 951020-.mra:No MameVersion
/media/fat/_Arcade/Strider -USA, B-Board 89624B-2-.mra:No MameVersion
/media/fat/_Arcade/The King of Dragons -World 910805-.mra:No MameVersion
/media/fat/_Arcade/Three Wonders -World 910520-.mra:No MameVersion
/media/fat/_Arcade/U.N. Squadron -USA-.mra:No MameVersion
/media/fat/_Arcade/Varth  Operation Thunderstorm -World 920714-.mra:No MameVersion
/media/fat/_Arcade/Willow -World-.mra:No MameVersion
```

jotego, 2020-09-09:
I won't correct it myself. I leave the issue open in case someone wants to add the information to the MRA files in https://github.com/jotego/jtbin

jotego, 2021-01-31:
Atrac17 and Chocobobo are working on a new MRA set for these cores. This is covered by their solution so I will close the issue here.

closed 2021-01-31 by jotego