jotego/jtcps #42: 1941 - right edge of screen (in landscape HDMI display) has weird defect

c0d3h4x0r, 2020-06-09:
Not sure if this is accurate of the original hardware or not, since I don't have original hardware to compare against.  But I've noticed that a few columns of pixels at the right edge of the screen don't appear to render correctly.

[image]

[image]

jotego, 2020-09-09:
I have tested it:
-It doesn't occur in MiST
-it occurs in MiSTer for analogue 15kHz output or HDMI (rotated or non rotated)
I'm going to investigate it

jotego, 2020-09-09:
It only happens if the DIP switch for flip is enabled both in MiST and MiSTer. I didn't see it in MiST because the flip was off by default.
This points to some problem in the scroll data DMA involving the flip bit.

jotego, 2020-09-10:
This is now fixed in commit 1b6327b7fc1201a27b7d390cb54178d6def1d08f
I'll create new RBF files. I'm closing the issue now.

jotego, 2020-12-07:
This happens again in commit https://github.com/jotego/jtcps1/commit/bac12ec630ee3a4161137427fa8bc1f7a2836994 after fixing this assignment:
```
-wire        tile_vs  = vrender1== flip ? 9'd17 : 9'd12; // Vertical start, use for SCR3
+wire        tile_vs  = vrender1 == (flip ? 9'd17 : 9'd12); // Vertical start, use for SCR3
```

M-Walrus, 2021-01-31:
Adding another example of this bug occurring in Varth:
https://youtu.be/A_6O4TvEd8E

Clouds and large boss ships exhibit this bug throughout the game.

jotego, 2021-02-01:
This is fixed now locally. I will update the binaries later (today).

closed 2021-02-01 by jotego