jotego/jtcps #45: Street Fighter II': Hyper Fighting (USA 921209)

mrfpga, 2020-06-27:
In 1P mode on the Balrog (Claw) stage there are graphical glitches in the center of the stage, where the women are dancing in the background. Thre is a horizontal stripe below their feet with random colours that are permanently changing.

EDIT: The problem occured on the MiSTer core, not sure about the other systems.

mrfpga, 2020-07-01:
@jotego 

Here is a video showcasing the bug, which is already present on Street Fighter II: Champion Edition:

image:
https://ibb.co/52Xc3Gs

video:
https://youtu.be/tHlIP2nwj3M?t=834

cracklejoy, 2020-07-22:
This is not a bug.  Those are the lights for the stage and have been there for in sf2,sf2ce and sf2hf.

closed 2020-07-23 by mrfpga