jotego/jtcps #49: CPS1 Core button mapping not working on SF2 games for X-Arcade TankStick

wiggy808, 2020-07-03:
@jotego 

Patreon here. 
I've submitted a ticket to X-Arcade techs, they indicated (as well as Classic Gaming Discord channel) that it was a core issue. 
The X-Arcade Tankstick is basically a keyboard encoder, so opening up notepad, the buttons input letters and keys from the keyboard. 

During button mapping of Player 1 on any of the CPS1 Street Fighter 2 series (SF2, SF2CE, etc. ), Player 2 does not have button mappings for kicks, only the 3 punches, while Player 1 has all 3 punches and kick buttons mapped and working. 

Couple things that might help:
1.) Final Fight works great for both players, as they each only need 3 buttons instead of 6, so I'm assuming that might help you key in on the issue some way. 
2.) The device does come up as keyboard on MiSTer when CPS1 loads, and same ID is detected when doing button mapping for both player 1 and player 2 because there is only 1 USB plug that is used to carry both players input. 

Let me know what I can do to map player 2, as the .map and .cfg input files are not ascii so i don't know of a way to just set them in a file manually. 

Thanks 
Wiggy

jotego, 2020-11-19:
Sorry for the long delay in response. When I first read this I thought someone else would reply with a solution. Sometimes it does happen!

Now that I read more, I think it may be indeed a problem with the core. My cores convert key presses to joystick movements and that is probably interfering with MiSTer firmware, which also may be trying to do that for your joystick.

I will release the second CPS 1.5 beta without the internal keyboard mapping so it will all be left to MiSTer firmware to handle. Let me know how that goes.

Apologies again for missing to reply promptly.

wiggy808, 2020-11-19:
No worries, I’d say this is probably solved no need to change anything. 

I was able to narrow down the problem (as it occurred on the SNES core as well), to xinput vs Dinput modes on my joystick. 

See my ticket here as it’s now solved as of yesterday so everything works on CPS1 and other cores as well.  

https://github.com/MiSTer-devel/Main_MiSTer/issues/276#issuecomment-730326217

jotego, 2020-11-20:
Thank you for your feedback. I always doubt about removing the keyboard mapping from the cores in case some users already use it. 
I will close the issue now.

closed 2020-11-20 by jotego