jotego/jtcps #51: MiSTer update_all removed Xinput support for CPS1

wiggy808, 2020-07-24:
Not sure what part of the update_all script affected this. 

I had an image that I added CPS1 cores too via Patreon. Unable to map the keys (other issue already open) for the X-Arcade Tankstick, I upgraded the PCB to the Tri-mode board so that it could be used as a DirectInput or Xinput or Keyboard Input (only option before upgrade). I tried both DirectInput and Xinput modes, all buttons for both players worked fine. 

After running an update_all, I'm now only able to use DirectInput mode for CPS1 cores, Xinput no longer allows me to map Player 2 buttons. Player 2 moves player 1.

jotego, 2020-07-24:
I don't see how this can be related. Let me try to get the author of update-all script into the conversation.

theypsilon-test, 2020-07-24:
Hi @wiggy808 , is this issue also happening with other arcade, computer, and console cores?

theypsilon, 2020-07-24:
Hi @wiggy808 , is this issue also happening with other arcade, computer, and console cores?

wiggy808, 2020-07-24:
No, but SF2 is the only core that requires 6 buttons per player. Even before I upgraded to Tri-Mode PCB, I was able to play Final Fight etc in keyboard mode on XArcade Stick. 

As long as I only have to map a few keys, any of the modes seem to work. It was just something I noticed after upgrading to the Tri-Mode PCB that I couldn’t map all the keys in Xinput mode, only in Directinput mode. 

So it’s definitely possible that other cores are affected, I just don’t have another core that requires 6 buttons per player to be a good test, so I opened up the issue with your CPS1 core. 

Lemme know how I can test this better to help narrow down the cause.

wiggy808, 2020-07-24:
Apologies didn’t hit reply all. 

No, but SF2 is the only core that requires 6 buttons per player. Even before I upgraded to Tri-Mode PCB, I was able to play Final Fight etc in keyboard mode on XArcade Stick. 

As long as I only have to map a few keys, any of the modes seem to work. It was just something I noticed after upgrading to the Tri-Mode PCB that I couldn’t map all the keys in Xinput mode, only in Directinput mode. 

So it’s definitely possible that other cores are affected, I just don’t have another core that requires 6 buttons per player to be a good test, so I opened up the issue with your CPS1 core. 

Lemme know how I can test this more to help narrow down the cause.

theypsilon, 2020-07-24:
Maybe you could try with SNES and Genesis cores, they have 6 button controllers.

wiggy808, 2020-07-24:
Good call. I added SF2 rom for SNES, same issue with Xinput selected on X-Arcade as I saw with CPS1 core and no problems with DirectInput selected for SNES SF2.

Unfortunately, I would have to assume XInput worked on SNES before the update_all for the SNES core as well, since I only had tested the CPS1 core just before and after the update_all (I primarily use the MiSTer for arcade). 

Couple thoughts. First, Not sure how to move this ticket to a MiSTer issue vice CPS1 issue like is now. Second, just to clarify how I’m doing the button mapping: 

1. I unplug X-Arcade (not necessary but for good measure)
2. Set Switch to Xinput mode on Xarcade
3. Plug x-Arcade back in. 
4. Reset MiSTer
5. Select OSD and Reset to map new joystick
6. Map joystick 1 starting with A button on joystick 1 
7. Redo #6 with A button and map all of player 2 joystick controls. 
8. Start SNES core —> SF2 
9. Bring up OSD menu
10. Select Map buttons for game using A button on joy1 
11. Repeat but using A button and map joy2

Using the exact steps above, when step 2 is on Xinput, when all this is done, Joystick 1 and Joystick 2 both move / control player 1 (like the CPS1 core). 
When step2 is on DirectInput, both joysticks work as player 1 and player 2 just fine and have all 6 buttons working as expected. 

Hope this helps, lemme know if I can test anything else.

theypsilon, 2020-07-24:
I think the better place to post this issue would be here: https://github.com/MiSTer-devel/Main_MiSTer/issues

update_all might have installed a newer version of Linux or Main_MiSTer binary through the main MiSTer-devel updater, and that newer version might have some problem with Xinput.

You will get better support there for sure.

wiggy808, 2020-07-26:
Will do, thanks!

jotego, 2020-08-05:
Closing it here then.

closed 2020-08-05 by jotego