jotego/jtcps #53: Options menu in Street Fighter Alpha (CPS Changer)

raziel2001au, 2020-08-27:
It's supposed to be possible to access the options menu which contains settings like difficulty in the CPS changer version of Street Fighter Alpha by pressing Start 2 + Pause at same time for 1 second. However, the pause implemented in the core is not the same as the pause button on the CPS changer. Even if it was it's not a well known feature so I'm wondering if it's possible to add some sort of menu option or shortcut in the core to access the settings menu of the game?

jotego, 2020-12-17:
I have implemented this request. Sorry for the long delay and thank you for letting me know about this. I had no idea.
I will release it tomorrow.

closed 2020-12-17 by jotego

raziel2001au, 2020-12-18:
Perfect, thanks jotego! Out of interest sake, what kind of solution did you go with in the end?

jotego, 2020-12-19:
I just did what you said. The unused coin button is used for the charger pause. This means you have the original system pause available.

raziel2001au, 2020-12-20:
Perfect, thanks!