jotego/jtcps #54: No diagonal inputs on gamepad in Final Fight 30th Anniversary Edition romhack

TLPD-AVW, 2020-09-26:
Final Fight 30th Anniversary Edition romhack does not allow to perform diagonal inputs on gamepad's dpad, making it impossible to move characters diagonally and greatly increasing the game difficulty. Keyboard is unaffected by this issue, allowing proper character movement in all 8 directions. I'm using the MRA [identical to the one provided by the hack authors](https://github.com/originalgrego/FinalFightAE-Installer/blob/master/ffightae.mra) with the exception of different rom paths. Other Final Fight versions are unaffected by this problem.

cracklejoy, 2020-09-29:
Experiencing the same issue

jotego, 2020-09-29:
Change this in the MRA:

from 
`<rom index="1"><part> 72 </part></rom>`
to
`<rom index="1"><part> 0 </part></rom>`

That will fix it.

closed 2020-09-29 by jotego