jotego/jtcps #55: CPS1 background layer missed in Three Wonders intro.

jegayon, 2020-09-29:
There seems to be a problem in the background layer in the introduction of Three Wonders:

[image]
[image]
[image]
I hope these screenshots help.

paulb-nl, 2020-09-29:
With Turbo enabled the backgrounds appear correctly.

jotego, 2020-09-30:
Excellent hint. Thank you!

jegayon, 2020-09-30:
Umm... I must have something wrong with my Mist core, because even with the Turbo option activated the issue remains....

jotego, 2020-10-01:
Turbo function doesn't work in MiST from the OSD.
I can confirm that enabling turbo fixes it in MiSTer.

jegayon, 2020-10-02:
Yes!!! I just recived my Mister today and I can confirm it too.

jotego, 2020-12-07:
I tested it today using the new SDRAM controller, in case this was a memory contention issue. But the result was the same: no background for no turbo, background for turbo.

I wonder if the game should actually be set at 12MHz CPU permanently. I will try modifying the MRA to set 12MHz.

For my reference, this is how the scenes should look like.

[image]
[image]
[image]

jotego, 2020-12-30:
It happens even if CPU speed is set at 12MHz in the MRA file. Maybe the DMA starts a bit too early in the line.

jotego, 2020-12-30:
Compiling with FORCE_GRAY does not show the backgrounds either, so it is not a missing palette problem.

jotego, 2021-01-31:
Six months later...

The speed measurements I have made for issue #33 gave the final hint. When I applied the first fixed to the DMA (speeding the core by 5%), the backgrounds showed up. The problem was that the core was running too slow. I have it fixed locally and I will release the binary update today.

Thank you for reporting this and sorry for the long wait.

closed 2021-01-31 by jotego