jotego/jtcps #56: CPS1 Willow sprites not fading out with background.

squidrpi, 2020-09-29:
Noticeable on Willow opening screen (below) showing the worlds. When the background fades out the sprites do not fade, just switch off. Possibly related to [#32]

[image]

closed 2020-10-03 by squidrpi

paulb-nl, 2020-10-04:
The sprites do fade with the background but it is difficult to see with the current brightness curve. The sprites do stay a few frames longer than the background.

Here is a video with the fade out going darker where it is easier to see:
[https://youtu.be/a_vxJwQQTuA](https://youtu.be/a_vxJwQQTuA)

jotego, 2021-01-18:
I have replaced the quick calculation method by a [look-up table](https://github.com/jotego/jtcps1/blob/master/cores/cps1/hdl/pal_lut.hex) obtained by solving the electrical circuit in the schematics. I believe this is as accurate as it gets. Please check it out, just update MiSTer to get the files.