jotego/jtcps #61: PCM sound missing in Street Fighter II series.

PepAlacant, 2020-11-26:
It lacks a sound on Chun-li Stage music. It seems a PCM is missing on the melody. It's a instrument that sounds almost all the time in the melody. Here is a sample of how it should sound. 

https://www.youtube.com/watch?v=H2dgVJy8jZM

jotego, 2020-11-26:
That's just an awesome finding. And quite an embarrassing one too. There is definitely a PCM channel missing. I will get it fixed.

PepAlacant, 2020-11-28:
It happens in other games too, lost or very low PCM channel. I suppose it's the same issue than in Street Fighter II. The King of Dragons loses percussions and other games! Thanks for your excellent work.

jotego, 2021-01-01:
The sound commands are issued in time, and there is a sound produced at the right time. But it is not the same sound.

The sound seems to be produced by ADPCM phrases **4** and **B**. The Z80 sends two commands very closed to each other for each phrase: 4, 4 and then B, B. The PCM module is requesting the right data and getting it correctly.

Parsing the SF2 OKI ROM directly and listening to sounds 4 and B, I find that they sound quite different from how they sound in the arcade (or at least in MAME). Even when parsed with the MAME C code.

I do not understand right now why these two sounds are played differently. Some ideas:
-Side effect of the second issued command with the same phrase
-Different response to saturation

jotego, 2021-01-02:
It was just the linear interpolator removing too much HF content. Replacing it with a full FIR x2 interpolator preserves the HF content, and the sample sounds as expected. Removing interpolators all together is also a valid option.

I will release an update today.

closed 2021-01-02 by jotego