jotego/jtcps #62: CPS15 build

gyurco, 2020-12-13:
I think some things are missing for a successful build:
cores/cps15/patrons/msg
jtdsp16 module (also referred as git@github.com url, not https://)

jotego, 2020-12-17:
Yes, I rushed a little bit with the public files. A commit was missing inside the Mister template in JTFRAME too...
I am not planning to publish JTDSP16 as I am a bit afraid of commercial companies using the code without complying with the license.
I can give you access to the private repository if you want to see it. I will publish it when I am done with CPS2.
However, you do not need it to build CPS15. You can just define the macro NOSOUND, or simply compile with **jtcore cps15 -q** and you will get the core compiled (with no sound, though).

gyurco, 2020-12-18:
For my purposes, sound is not required. I just want to try the new EEPROM saving feature in the firmware.

jotego, 2020-12-19:
It should compile correctly now. I fixed the missing commits.
This is the kind of thing I want to prevent (or at least delay):
https://www.ebay.com/itm/Analogue-NT-NES-24K-Gold-Plated-Factory-Sealed-Brand-New-System-9-out-of-10-/193258367366?_trksid=p2349624.m46890.l49292

jotego, 2020-12-19:
About the EEPROM, the module I use for storing the values in the core already has a secondary set of ports there in place. The idea is to connect that to whatever saves to the SD. If you get that done, it would be awesome!

gyurco, 2020-12-19:
With a parallel interface for jteeprom, it would be a piece of cake. But I don't see that, what should I look for?

Yeah, you cannot know what was used by Analogue. Also it was very interesting when Ijor released his FX68K CPU, their Genesis suddenly appeared.

jotego, 2020-12-19:
I have connected the lines in jteeprom and pushed it.
Please update CPS15 with submodules and look for this line: https://github.com/jotego/jtcps1/blob/344ec615ca12e35470ecc473df793e886ec99bcb/cores/cps15/hdl/jtcps15_game.v#L315

gyurco, 2020-12-19:
I see now. A dump_we would be also needed to restore the EEPROM at core boot (I've implemented it by sending a .RAM file via ioctl_index=1 just after the .ROM).

gyurco, 2020-12-19:
Btw, which branch of jtframe should I use? I see the branch for cps15 is diverged from master, where the data_io change is already committed. Are you planning to merge jtframe/sdram back to master?

gyurco, 2020-12-19:
I've sent a PR to jteeprom. I have the necessary jtframe changes also, but in the master branch, which doesn't compile with cps15. So I wait with that.

gyurco, 2020-12-19:
Another problem that without the sound board, no controls at all, thus I cannot enter service mode and try to change some settings :)

gyurco, 2020-12-19:
It's the jtframe change for master:
https://github.com/gyurco/jtframe/commit/0d855969fe2864ad25d3e02fa4cc478a9f902a29
I won't send a PR until the dust settles on the repos, or you can integrate it in your own way. The cps15 change itself is very minimal.

jotego, 2020-12-19:
Thank you. Let me merge the changes in master to the SDRAM branch and check that it compiles correctly.

jotego, 2020-12-19:
It has taken a couple of trials but if you update CPS1.5 and run a submodule update, it should all work. I cannot still update the master branch of JTFRAME as the SDRAM interface has changed, and I still do not consider it stable.
Could you try your change on these files?

gyurco, 2020-12-19:
Looks like something is lost in the sdram branch, as USEWAITN is no longer in jtframe_z80_romwait.

gyurco, 2020-12-19:
...and I only get a black screen for every games.

jotego, 2020-12-19:
It's strange... I have just checked the commits in Github, and they are correct.
I tried compiling again a moment ago, and it well. I wonder whether you missed running the _submodule update_ command.

gyurco, 2020-12-19:
Now OK...for some reason I didn't get the two new commits in the cps15 repo.
Unfortunately only 00's are written in the RAM file. I have to check if the error is on the serial or the parallel side of the EEPROM.

gyurco, 2020-12-20:
It works finally! 
I sent you the PRs (submodule update is not included), and a new firmware via email.

jotego, 2020-12-20:
Just tested it here. It's pure joy!
Let me know here when you publish the firmware. I will release the updated core then.
Sound is still noisy with the sigma-delta output of MiST. I will be lowering the system frequency to 48MHz next. That may help with the sound noise too.
I may refactor the code a bit. Maybe push it to jtframe_dwnld or the core-specific _dwnld/prom_we module.
Thanks a lot.

gyurco, 2020-12-20:
Great! I've just released the new firmware (201220). Note that it also has 64-bit status word support, too (Street Fighter should use it  - it's broken currently).
I've tried to make this change as simple as possible, and it should be backwards compatible with existing cores, too (if no CORE_NVRAM_SIZE is defined, that the game interface is unchanged). But of course, it's your code, and it must fit to your taste, too.

48 MHz SDRAM clock for CPS1? Won't be too slow? Isn't it possible to keep 96MHz at least for MiST/SiDi? Maybe if the core and the sdram clocks can be separated (SDRAM controller can be 48 or 96, but all other parts are only 48, as it's done in many other cores), then it'll be easy to do it.

gyurco, 2020-12-23:
The noisy sound on MiST couldn't be because of the overall volume level is very low? Just compare it with CPS1, CPS15 is very silent.
Btw, Merry Christmas!

jotego, 2020-12-23:
Merry Christmas!

Yes, the volume is low but there is a lot of HF content that saturates the sigma-delta modulator easily. Reducing the frequency of the modulator helps in avoiding saturation. Having an integer relationship between the DAC clock and the sample rate also prevents noise and when I fixed that I got rid of horrible noise. But it isn't clean yet. I think messing a bit with the clock enable of the DAC may fix it.

This DAC output is for me one design flaw of all homebrew FPGA platforms. It takes sigma-delta modulation for granted, simplifying it to great lengths. On MiSTer it is possible to connect an external DAC and bypass the 1-bit modulator. The difference is abysmal.

jotego, 2020-12-30:
ioctl_index in data_io does not have a reset value, which kills my sims. I tried setting it to 8'd0 at SPI_SS2 posedge but that creates problems in the actual core compilation. Games don't boot up if I do it. I do not understand why. Does it make sense to you? @gyurco

gyurco, 2020-12-30:
You can use ioctl_download/ioctl_upload to determine if it's valid. If they're 1, then the index is surely valid.

gyurco, 2020-12-30:
Resetting it with SPI_SS2 will reset it every time the ARM stops using the SPI.

closed 2021-01-17 by jotego