jotego/jtcps #66: CPS 1.5 halts and odd behaviour

rarcos, 2020-12-20:
These crashes appear in release 201211.

There are random crashes in gameplay in both games:

[image]
[image]
[image]

Also, the games crash always at the same part in attract mode:

[image]
[image]

Also, Cadillacs and Dinosaurs shows this behaviour: enter the first building, defeat all enemies, but don't get close to the window and the the time run out, when the player respawns the sprite will become unstable and then disappear, a few seconds later, the next scene will start. I'll try to capture footage from this if I can reproduce it again.

jotego, 2020-12-20:
I think the issue is caused by excessive VDD ripple when all SDRAM banks are accessed in parallel. The new memory controller does that in order to improve data throughput. 

CPS1.5 with 4-bank interleaving had >700mVpp ripple
CPS1 with 1-bank access had <200mV ripple

[image]
[image]

jotego, 2021-01-01:
Please try today's update and fill the poll here:
https://www.patreon.com/posts/45683205

closed 2021-01-01 by jotego

movisman, 2021-01-01:
Thanks so much for the tireless work on this particular issue.

I will update later on and fill the poll accordingly.

Your Patreon post mentions "I know that at least two SDRAM module builders are going to try to improve the modules." - are these separate improvements from what Alexey mentioned on a live chat he took part in the other night (https://youtu.be/sVTMWX0WUJk), or are the improvements linked? Here is the quote from Alexey:

[image]

Just curious if the proposed/future revision Alexey mentions has any relation to the improvements the builders are trying to implement into their modules.

Happy New Year and all the best for 2021.

Thanks!

rarcos, 2021-01-02:
Thanks for your hard work. Latest ralease is stable for me.