jotego/jtcps #71: Sound issue in Street Fighter II series

nickcsbr, 2021-01-06:
Hello,

Since the last update I have noticed that the ADPCM samples in Street Fighter II series are sounding very low quality for the CPS1 core. The hit sounds, the airplane sound and specially the voices are the most noticeable ones. Please let me know if any other information is needed and keep up the amazing work! Thank you.

SegaSnatcher, 2021-01-09:
I noticed this too, glad it wasn't just me.

SegaSnatcher, 2021-01-11:
Final Fight sample audio also sounds worse now.

jotego, 2021-01-12:
I am implementing the original on-board filter and more precise upsampling filters for the next update. But, all the signal flow outside the core is out of my control. The different filters and D/A converters will affect the sound too. This is especially true if you use the cheap analogue output in MiSTer. Then there is another problem: many people only remember MAME, not the real sound.

Here is the frequency response of the on-board filter for the PCM channel. I hope you find this more to your liking when I finish it.

[image]

jotego, 2021-01-13:
I think this is solved now in commit https://github.com/jotego/jtcps1/commit/6e1934ac39cf118277dd936cc3d59d75383904aa
You can hear it in the attached file. I will release the update on Friday.

[final fight.mp3.zip](https://github.com/jotego/jtcps1/files/5810099/final.fight.mp3.zip)

closed 2021-01-13 by jotego

nickcsbr, 2021-01-13:
> I think this is solved now in commit [6e1934a](https://github.com/jotego/jtcps1/commit/6e1934ac39cf118277dd936cc3d59d75383904aa)
> You can hear it in the attached file. I will release the update on Friday.
> 
> [final fight.mp3.zip](https://github.com/jotego/jtcps1/files/5810099/final.fight.mp3.zip)

Perfect! Thank you for your incredible work!