jotego/jtcps #72: Sprite glitch in Strider

Hewhoisred, 2021-01-11:
This may be related to the star field bug.  It's reproducible.  You'll see it all over the game when ascending/descending slopes.
[image]

Hewhoisred, 2021-01-11:
I've noticed the black garbage seems to happen right where the sword graphic comes out.

Hewhoisred, 2021-01-11:
this may actually be correct behavior, the same glitch happens in MAME

Hewhoisred, 2021-01-11:
correct arcade behavior, disregard

closed 2021-01-11 by Hewhoisred

jotego, 2021-01-11:
I did check this a long time and came to the same conclusion. It may be ok on some different game version. It is possible to hack the ROM to delete that pixel. This could actually be an error in the ROM dumping.