jotego/jtcps #73: Controller issue in Final Fight

pacoarcade, 2021-01-14:
In Final Fight, when I press 'left' with player 1 controller the character moves like if I were pressing Left + Up. This did not happen in previous releases. 

Reconnecting and reconfiguring the joystick doesn't fix this issue. I'm using a Daemonbite Saturn to USB converter that works perfectly with other jtcps1 games.

jotego, 2021-01-14:
I have heard of similar things before. I think this gets solved by deleting the old config files. I think you can find them in the config folder of MiSTer. The name is either that of the MRA file or the ROM file associated with the MRA.

This is a MiSTer issue, as far as I know, not related to the core itself.

closed 2021-01-14 by jotego

pacoarcade, 2021-01-14:
Tried that deleting /media/fat/ffigh*, no success even powering off/on. Final Fight -Japan- version from 'alternative' works perfect.

jotego, 2021-01-14:
I suppose you meant /media/fat/config/ffight*

I suggest you ask in either my [Discord server](https://discord.gg/yA2N4V2y) or in Antonio Villena's [telegram group](https://t.me/misterfpga). I really don't think the issue is in the core.

pacoarcade, 2021-01-14:
That's right, /media/fat/config/ffight* . Anyway, Final Fight japan works great!

Thanks for the invitation, I've registered to Discord and verified my account but it doesn't work with my browser. I'd tried to join telegram some time ago but it requires a smartphone to sign up :(