jotego/jtcps #75: CPS1 ADPCM drums? (sf2*, ffight*)

ItsAtomico, 2021-01-16:
I might be in the wrong here, but I believe there's is some odd artifacting going on in the near the end of the kick drum sound on the ADPCM music for both Final Fight and the SF2 series. 

It's way too trebly, instead of being a dull "thump". Also seems like it has metallic/tinny sound afterwards the proper drum.

So yeah, I'm try to describe a sound in a foreign language, based on how  I _remember_ it from over a decade ago. Sorry about that and thanks for all your hard work.

MJY71, 2021-01-16:
To me, it sounds like a tambourine is taped to the back of a bass drum whenever the kick drum hits. There is a ringing/jingling that should not be there.

ItsAtomico, 2021-01-16:
Here's a few sound test codes for ffight: 

SFX:
0D is the knockdown sound. 6D appears to be the same one.
26 is the fast running steps from Sodom and maybe the Andores.

Stage music starts at address 40 and goes from there:
40 is Slums. 
41 is the basement with the fat dudes. Drums are even more prominent there

ItsAtomico, 2021-01-17:
Fixed in jtcps1_20210117.

Thanks JT! Great work as always.

closed 2021-01-17 by ItsAtomico

SegaSnatcher, 2021-01-17:
> Fixed in jtcps1_20210117.
> 
> Thanks JT! Great work as always.

Where is the jtcps1_20210117 release at?

jotego, 2021-01-18:
I uploaded it to the server last night. Just run the updater.