jotego/jtcps #78: CPS1: 1941 very low FX volume option still over powers music.

SegaSnatcher, 2021-01-19:
I feel even the lowest volume FX option might be a bit too loud for those that want a more balanced audio between FX and FM.  

Would be nice if very low was adjusted slightly lower so the FM music could be heard more.  This is with the filter setting off.  

Here is an example of what I like to hear.

https://www.youtube.com/watch?v=rEqFyEjePZA

PhantombrainM, 2021-01-31:
Your Video says "Ver mame0.37b16".

The behavior of the music fx balance should be compared to a real pcb. If that mismatchs a issues ticket  should be created. 

Otherwise it's more of a feature request than an issue. 
JT cores already allows to disable FX completly, if you just want to listen to the music. Not sure if it needs even more options.

jotego, 2021-02-01:
I want to make a test for sound balance on the PCB, but this is going to take several weeks I suppose.
Once I have that information I will re-centre the OSD values around the measured value. I will make the "very low" option really sound very low.

SegaSnatcher, 2021-02-01:
Awesome!!!

SegaSnatcher, 2021-02-27:
Thanks for making FX lowest volume lower, but it seems all volume levels are now lower as a result.  Was this intentional?

jotego, 2021-02-27:
Yes, I turned everything down a notch. Let us leave it like that until I can make the measurements.

closed 2022-04-17 by jotego