jotego/jtcps #79: "Original Filter" triggers noise artifacts (analog audio) MISTer

PhantombrainM, 2021-01-31:
If you load up Carrier Air Wing and let the first attraction cycle pass there is a remaining permanent sound with the following conditions:

-> MISTER CORE VOLUME in GUI Core Settings -> one step down 
-> Original Filter : ON (OFF -> Voice gone)
-> FX Volume : LOW (other settings change pitch of sound)
-> FX: ON (OFF -> Voice gone)
-> FM: ON OFF -> Not affected
-> MISTer general Volume change F11/F12 (Full and one step lower changes pitch of the voice)

I can try to make a video if necessary.

paulb-nl, 2021-01-31:
This is caused by the audio DAC of MiSTer. SNES also has this issue.  Using Optical or HDMI audio extractor there is no issue.

jotego, 2021-01-31:
Like Paul says, this is artifact created by the cheap built-in DAC. There are techniques to minimize these problems but there is only so much you can do.
My recommendation since several years ago is to stop using cheap 1-bit sigma-feta converters and have a proper DAC connected to the FPGA.
MiSTer allows that by means of the optical output or the HDMI output. 
I hope future FPGA systems for retro have a proper DAC installed on board. 

I can try and add options to mess with the built-in DAC, like:

-reduce signal amplitude
-add dithering 
-filter out low frequencies 

But I think it's better if users buy an 8€ DAC, honestly. 

Let me know what you think.

PhantombrainM, 2021-01-31:
I hadn't tried SNES, but I noticed it with the 48Mhz version already with Punisher. In the 96Mhz version it was gone again.

If it's a hardware problem with the Mister, then I'm not in favour of a workaround. Who knows where the journey will take us. That would then have to be removed again in a future hardware generation or revision, so not purposeful. 

 I'll look into an external converter or I'll try to grab HDMI audio.
 
 I think the ticket can be closed with this.

PhantombrainM, 2021-01-31:
Closed. As discussed.

closed 2021-01-31 by PhantombrainM