jotego/jtcps #80: Missplaced pixels on Street Fighter 2's tittle logo.

PepAlacant, 2021-02-01:
[image]

Just one or two pixels moved from their place. Happens in mist/mister. I didn't noticed the problem in other games by now.

jotego, 2021-02-01:
This is most likely a problem in the ROM file. I have seen this kind of problem in two other games. They are due to incorrect ROM dumps.

ItsAtomico, 2021-02-01:
Similar issue in SFZCH:

[image]
[image]
[image]
[image]

Weirdly enough it only happens when Guy's standing on the left side of the screen. When he's on the right the sprite doesn't glitch.

jotego, 2021-02-01:
Well spotted. I have caught the problem. I will release an update today.

closed 2021-02-01 by jotego