jotego/jtcps #84: [MiSTer] CPS 1 : 1941 - Counter Attack (Japan) -  reverse display for rom check part on HDMI

wwark, 2021-02-13:
This case is present just on HDMI, the case is ok on CRT.

I took screenshot with the Mister (option win+printscreen)
Rom Check
[image]
In game
[image]

We saw trhe rom check is inverted compared during the game.

Thanks,

jotego, 2021-02-13:
The screen orientation is controlled by the game. It seems like the game doesn't apply the flip setting until later. 
I'm afraid there isn't anything I can do.

closed 2021-02-13 by jotego

wwark, 2021-02-14:
Ok on Mame the behavior is not reverse on my LCD screen. Is Mame not have the correct behavior is this case ?
and also this issue is not present on CRT.

Thanks,

jotego, 2021-02-15:
Set the flip setting in MAME and you will see it happening there too. 

For the CRT test, you must have run it with a different flip setting.

wwark, 2021-02-15:
ok if i set flip Mame i saw the reverse display but it is inverted between the mister core and Mame.
Thanks no need to check more on this case.