jotego/jtcps #87: Street Fighter Alpha (CPS2) Ryu's ending

jotego, 2021-03-02:
Ryu's ending screen looks black instead of showing an orange sky.

The sky is in layer SCR2, but SCR3 is configured to have priority over it, and it is all drawn with ink zero, which is not transparent. There must be some bit in the CPS-A/B chip to configure this, but we don't know it.

This is how it looks like:

[image]

And this is how it should look:

[image]

jotego, 2021-03-02:
This seems to be a ROM mapping problem. CPS2 is assumed to cover all ROM space for the SCR banks, but in this case SCR3 may be pointing to an unmapped region, and thus should be read as FF's (and be transparent)

Chocobobobo, 2021-03-04:
Just checked with my good friend Raph, who owns the blanka version of the PCB

[image]

paulb-nl, 2021-03-04:
Street Fighter Zero PCB:
https://youtu.be/P-zc4VzSZgU?t=396

https://youtu.be/qRfPZ-rqujY?t=817

Chocobobobo, 2021-03-04:
Yes, I watched every single video with it on YT. But with the poor image quality, it feels like some are using super eagle filter (meaning emulators are being used). This is why I am just trusting people I know, owning actual hardware (and streaming it for me to check).
But thanks Paul :)

closed 2021-03-09 by jotego