jotego/jtcps #91: Graphical glitches on right side of screen (jtcps1_20210307)

MJY71, 2021-03-08:
With release 20210307, there is a single pixel-wide column area that extends just beyond the right border of the screen where objects in certain layers scrolling to the right of the screen or objects that are already just off-screen show up as graphical garbage in certain places.

Screenshots of Street Fighter Alpha (CPS Changer) shown, but it happens in other games as well.

[image]

[image]

PepAlacant, 2021-03-09:
There are other glitches that appear at Street Fighter II, II' and Turbo intro.
https://user-images.githubusercontent.com/75079781/110405822-7c54c980-8081-11eb-86a8-bf6005862283.MOV

jotego, 2021-03-09:
Fixed in 210309

closed 2021-03-09 by jotego