jotego/jtcps #94: Willow - Dip switches does not work properly

thehughhefner, 2021-03-30:
Hi,
I was playing through willow (world) and I've noticed that the dip switches are broken. Toggling one dip switch can affect other dip switches. For example, when increasing health, it can turn off other things. 

Also, the level dip switch doesn't work either, setting it to level 6 won't take me to that level.

atrac17, 2021-08-04:
These are the proper dips for Willow. Those others come from mame. Try placing this in your mra for [willow] primary mra.

```
    <switches base="23" default="3f,1f,de">
        <!-- DSWA -->
        <dip name="Coin A" bits="0,2" ids="2/1,4/1,3/1,2/1,1/4,1/3,1/2,1/1"/>
        <dip name="Coin B" bits="3,5" ids="2/1,4/1,3/1,2/1,1/4,1/3,1/2,1/1"/>
        <dip name="Cabinet" bits="6" ids="Upright 2 Players,Upright 1 Player"/>
        <!-- DSWB -->
        <dip name="Difficulty" bits="8,10" ids="8,7,6,5,1,2,3,4"/>
        <dip name="Nando Speed" bits="11,12" ids="Very Fast,Fast,Slow,Normal"/>
        <dip name="Power Ups" bits="15" ids="On,Off"/>
        <!-- DSWC -->
        <dip name="Lives" bits="16,17" ids="4,3,1,2"/>
        <dip name="Vitality" bits="18,19" ids="2,5,4,3"/>
        <dip name="Flip Screen" bits="20" ids="On,Off"/>
        <dip name="Demo Sounds" bits="21" ids="Off,On"/>
        <dip name="Allow Continue" bits="22" ids="No,Yes"/>
        <dip name="Game Mode" bits="23" ids="Test,Game"/>
    </switches>
```

jotego, 2021-11-21:
I have copied the DIP switch list provided by Atract17 to the MRA files.
The confusion came because Willow DIP switches have a different meaning if you turn on the service mode while playing -and don't reset the game. But there is no way right now to show an alternative DIP switch menu in MiSTer. So we have to delete the cheat options provided in this other mode.

closed 2021-11-21 by jotego