jotego/jtcps #98: Street Fighter 2 Champion Edition won't launch

ryosanisback, 2021-06-04:
Hi,

Mister and everything is up to date (with update_all script) but Street Fighter 2 Champion Edition won't launch.

Have this problem only with this game.

Regards

jotego, 2021-06-05:
Hi, it works for me. Try deleting old configuration files. Go to the config folder in your MiSTer SD card. Then look for files named sf2ce and delete them. There's a good chance that solves your problem.

ryosanisback, 2021-06-05:
Hi thanks for your answer.

Found the solution the rom downloaded with the update_all script wasn't good (too small, not complete, with no doubt a internet connexion problem).

I replace the rom zip into mame game folder with a good one and now it works.

Best regards,

Julien

Télécharger Outlook pour Android<https://aka.ms/ghei36>

ryosanisback, 2021-06-05:
Re hi :)

I just become a patreon beta tester :)

For CPS 2 i saw the mame version for rom is 0.229.

But I can't found game in this version.

Can I use rom game from 0.217 set ?

Without problem ?

Best regards,

Julien

Télécharger Outlook pour Android<https://aka.ms/ghei36>

jotego, 2021-06-06:
Yes, most of the time you can use older MAME ROM versions. However, you don't need to look for the ROMs if you use the update_all script. As far as I know that script will download the ROMs for you.

closed 2021-06-09 by jotego