jotego/jtcps #99: In "Muscle Bomber Duo Ultimate Team Battle" 3rd and 4th players can't jump

13r, 2021-06-06:
I mapped several gamepads but always the 3rd and 4th player have the "jump" button inoperative. In the OSD i can map the button fine in every pad, but in the game it doesn't work. I even switched gamepads and it happen always with the 3rd and 4th player.

I tried other CPS and CPS 2 games of 3 and 4 players where they use with 3 or more buttons and they seem to work fine (for example "Dungeon and Dragons Tower of Doom"). This seems to be the only game with this problem.

jotego, 2021-06-08:
That's indeed the case. Let me look into it.

13r, 2021-06-08:
Ok, thanks for confirming it (i was a little paranoid about some kind of configuration issue on my mister). Take all the time you need ;).

jotego, 2021-06-09:
Fixed in f6c327f
Run the updater to get the file

closed 2021-06-09 by jotego