jotego/jtcps #100: SF2WW Blanka headbutt bug sound

mikey1964, 2021-06-10:
play as guile get to blanka stage block everything at some  point blanka will try to headbutt you and will do it 2 times in a row and each time the dying sound plays not sure this happens in a real arcade board  .  also not sure if this is normal on a real arcade also as the plane moves the portraits flicker

mikey1964, 2021-06-10:
here is a video https://gofile.io/d/RMOFbA

jotego, 2021-06-11:
Hi,

I think your RBF may be a bad one. I made a bad update last week and had to delete it. The problem we fave is that the updater doesn't delete files, only updates. So I think you will have to manually delete the core.

I am preparing changes to the update approach to prevent these situations but it will still take a while until we have them ready. A first step will be announced today.

Thank you for reporting it.

mikey1964, 2021-06-11:
ok i will delete the RBF by hand and then let the update download it again. i  will test again and let you know if it made a difference

mikey1964, 2021-06-11:
i remove files manually   jtcps1_20210604.rbf - size 2.87 MB (3,013,356 bytes)
                                        jtcps15_20210604.rbf - size 2.91 MB (3,059,688 bytes)
is there anything else i need to remove. update install these files
jtcps1_20210604.rbf - 2.87 MB (3,013,356 bytes)
jtcps15_20210604.rbf - 2.91 MB (3,059,688 bytes)

 it made no difference
i got blanka to do the headbutt 3 times, in one set i got elephant sounds other 2 sets the sound had distortion and sound drop got like a robot like sound.

mikey1964, 2021-12-23:
so iam guessing you fix this JT? i let blanka do his headbutt  i see nothing change still getting different sounds

jotego, 2021-12-23:
Let me check again

mikey1964, 2021-12-23:
which rbf are you using

mikey1964, 2021-12-23:
iam using jtcps1_20211217

jotego, 2022-04-17:
I think this is probably a bug in the original. I'll close it until new evidence pops up.

closed 2022-04-17 by jotego

mikey1964, 2022-06-14:
i have not been able to find a arcade machine to test this out but atleast i can confirm this happens in mame  also so its not my de10 or bad files . if i find a arcade machine i will let you know if it does the same thing