jotego/jtcps #101: X-Men Vs. Street Fighter (Japan 961023) - Graphical issues

wwark, 2021-07-02:
You will line glitches on the screenshot below - produce with the 1st public release version.
[image]
Thanks,

jotego, 2022-01-07:
Technical notes:
-There are raster effects occuring close to the top of the screen. It is possible to see a blank in the background. Its height is not fixed and shrinks when the characters jump

jotego, 2022-07-03:
After recording a long capture of the attract mode, I was able to see it. It's around the street light, in the building. This video captures the [instant](https://youtu.be/BUo3GhcZbsY?t=1212).

We have seen similar ones that occurred in the original hardware. But it could also be a core problem as the raster counter could be wrong by one, or something silly like that. 

But I need an easier way to reproduce the bug. Do you have an easy way to get to that screen?

wwark, 2022-07-03:
Thank you to checking on this issue.
Maybe another way to reproduce it is to start a new game and hope this stage is proposed.
Maybe if you select some character it appears more frequently.
If the issue is similar on real hardware maybe the core do the right thing.
Do you have also a record of real hardware to check both behaviors ?
Thanks again !

star69rem, 2022-08-07:
This problem is in MAME, too, so there's obviously something off about people's assumptions about the way the hardware works.