jotego/jtcps #103: CPS1: Street Fighter II' Hyper Fighting -World 921209-.mra - Pause and Coin always mapped to R / L buttons regardless of configuration change

jdalmanza, 2021-07-06:
I can change the button mappings manually within the interface but no matter what I assign to L and R they always function as Coin and Pause. 

In my case I am using a SNES style gamepad. I can map all os the 6 input buttons to the buttons I want but L and R are always Coin and Pause. this is the only game that exhibits this behavior of the roms that I have tried on both CPS1 and CPS2 cores, I can freely map and replace coin and pause on just about every other rom, just not on SF2:HF

jotego, 2021-07-06:
Try going through the controller setup in the main MiSTer menu first (before loading the core)

jdalmanza, 2021-07-12:
I tried a different controller, HORI PS3 fighting pad and did not see the issue I described. Then, I took your advice and re-mapped the original SNES style controller and after that I did not see the issue anymore on this or any of the other ROMs.

closed 2021-07-12 by jdalmanza