jotego/jtcps #104: 3p and 3k joypad binding

buvk, 2021-07-14:
I was just curious whether it would be possible to implement a feature where you can bind 3p and 3k to a single button in games like ssf2t.

Thanks!

jotego, 2021-07-14:
Hi, maybe MiSTer has a generic way to map controls like that. I don't think I'll modify the core for that as it can be better done by the firmware layer (ie MiSTer).

Maybe ask in the MiSTer forum or MiSTer discord. 

Sorry for shooting the ball forward.

closed 2021-07-14 by jotego

buvk, 2022-07-03:
@jotego sorry to bug you about this again, but it doesn't look like 3x button bindings are going to be added to the firmware layer (per https://github.com/MiSTer-devel/Main_MiSTer/issues/427).  That being said, is there any chance you'd be willing to implement this functionality into your cores?

jotego, 2022-07-04:
I actually added a 2x button binding in the 1943 core. Not sure if anyone ever noticed it, though.
It's quite a mess to do it in the core, to be honest. And I still think this sould be done at the firmware level, which is where the inputs are parsed and handled to the core. I am sorry Sorgelig doesn't want to do it. Maybe one of the upcoming MiSTer altenatives will be provide this functionality.