jotego/jtcps #108: CPS-2 - Darkstalkers: The Night Warriors - Graphics Glitching

M-Walrus, 2021-10-31:
Elements of backgrounds, UI and character sprites all exhibit glitchy behavior when playing through the game.  These glitches happen for a frame and all elements have similar behavior where a block will be missing or misplaced.  This can be difficult to observe because of how briefly it flashes, but I have seen it occur on multiple stages while playing the game.  Does not seem to effect the intro, character select, or transitions between rounds.

CPS-2 core version where this is observed: 20211024

In the attached screenshot you can see it in the top-left corner where part of the UI is missing for a frame, which should be overlaid on top of the character artwork and framing it fully, but instead is missing the bottom half.
[image]

Another example of glitching UI element is timestamped here in my recent video for the core/game. Look a couple seconds past where this starts when "3 HIT COMBO" comes on the screen.  The I in HIT flickers out and then part of the word COMBO disappears with the B getting moved next to the T in HIT:
https://youtu.be/ouYL4MP_wNE?t=679

thehughhefner, 2021-10-31:
The Night Warriors is full of visual glitches. 

This happens on real hardware as well.

jotego, 2021-11-01:
Another example copied from Discord:
https://youtu.be/ouYL4MP_wNE?t=657

> You can see a large chunk of the background go missing on the right side for a frame, and part of it appears below Victor as he is falling: then if you go frame by frame as Victor his hitting the floor afterwards, you can see most of Rikuo's portrait by his life bar disappear
a few frames later if you observe Victor while he is bouncing off the ground, you can see large portions of his character sprite missing also
I noticed the player character sprites on multiple stages doing this, with large chunks missing from them for a frame.
same for background art on various stages

Still not sure whether this is genuine or a core problem. It is not so easy to spot it as it happens for a split second

M-Walrus, 2021-11-03:
Some videos were posted on discord from original CPS2 hardware with a darksoft multi, and the observations I made from those are:

turbo1:
-clean. no UI, character sprite or background issues.  MiSTer core seems to match this perfectly.

turbo2: 
-frequent UI corruption including character portraits flickering out for a frame.
-some background corruption. foreground light pole on jon talbain stage was most notable. no background objects randomly moving around the stage as was observed on MiSTer.
-minimal/no character sprite corruption (saw a single line on demitri for a frame, but thats it- no large chunks)

turbo3:
-definite character sprite corruption. half of demitri's sprite was missing for a frame during jon talbain fight.

so currently the only things in question in the MiSTer CPS2 core compared to what I've seen in the real hardware videos are:

-a regular occurrence of character sprites missing chunks

-stage objects moving to different places on the screen for a frame.  In my video linked above you can see the lightpole in the foreground on jon talbain's stage move to a different spot on the screen for a frame just like the background machinery did on Victor's stage.

thehughhefner, 2021-11-03:
This could be due to two things
- There are variations between real hardware. As mentioned before, Darkstalkers is in general a visual buggy game.
- DMA timing needs to be adjusted.  Didn't the mapping of the CPS-A confirm it was correct?

M-Walrus, 2021-11-03:
This may be coincidence, but the three instances where I've noted screen objects being drawn in the incorrect location for a frame, have always been shifted to the left on the screen.
The "3 HIT COMBO" text where the B was shifted to the left against the T, machinery on Victor's stage appeared on the ground (far) to the left of its original position, and the light pole on Jon Talbain's stage also moved to the left from its original position.

We have only seen a 5 minute video of Turbo 2 on original CPS-2 hardware so far, so I think a longer video would help to be certain here.

thehughhefner, 2021-11-03:
I'm starting to think more about what you are seeing @M-Walrus is also in real hardware. Darkstalkers is one of many where there are absolute random visual glitches that appear on screen for a frame. Also, it is very very hard to compare real hardware vs "emulator" since lots of things going on at the same time. Its a fighting game after all.

M-Walrus, 2022-04-02:
Closing this.  Last tests suggested it possibly occurred more frequently on the core than on a real PCB, but I have no way to verify this and do not want to have this report lingering around any longer. Possibly someone who especially likes this one can verify it closely to the original PCB performance at a later point.

closed 2022-04-02 by M-Walrus