jotego/jtcps #110: CPS2 - Super Street Fighter II X Grand Master Challenge (Japan 940223) - Reset when Selecting O. Dee Jay

springah, 2021-12-08:
When playing any of the SSF2 Grand Master Challenge MRAs (except Phoenix Edition) and selecting Original Dee Jay as a character (highlight Dee Jay, press Light Punch, then Down Down Up Up Heavy Punch), the game resets as the match is about to begin.

Apparently a known issue, but unable to find it reported here.

Tuxosaurus, 2022-01-10:
I can confirm, also with version 940311 and the freeplay hacks.

Here it is on video: https://drive.google.com/file/d/1TtECqYrBSW9Gn90MnyltJ9dHLhaM-E5B/view?usp=sharing

To reproduce, select DeeJay and then input Down Down Up Up + Light Punch (you'll hear him say "alright!")

springah, 2022-04-02:
Confirmed working in Phoenix Edition.

jotego, 2022-05-07:
Still happens in eee00ef after improving jtframe_romrq

jotego, 2022-07-03:
The decrypter address range was wrong. Fixed in bf5a34e

commit bf5a34e3 (2022-07-03): Fixes #110 and probably #125
closed 2022-07-03 by jotego