jotego/jtcps #115: CPS1: KOD Only one credit starts the game

maxxsun, 2021-12-22:
jtcps1_20211002     only one credit can start the game
https://www.youtube.com/watch?v=2FEMQDh0h6s

jotego, 2021-12-22:
The problem is that if you enter more than one credit and press 1P, the credit counter will go down but the game won't start. This repeats until you have 1 credit, then the game will start.

Items tested without effect:
-DIP switch settings (tested in MAME for unused pins)
-Disabling dial_cs
-Disabling ppu2_cs via 'h1Cx
-Disabling joy_cs
-Making the CPS-B ID 16'hFFFF

jotego, 2022-01-12:
Enabling the turbo mode in the OSD fixes the problem. The CPU speed seems to be wrongly assigned to 10MHz in this game. It should be 12MHz.

closed 2022-01-12 by jotego

jotego, 2022-01-12:
The files are updated now in JTBIN. You will need to update both the RBD and the MRA to get the correct behaviour. Thanks for reporting this :-)

commit 1c413628 (2022-01-12): Fixes CPU speed stuck at 10MHz in some games. Fixes #115