jotego/jtcps #116: CPS2 - Alien Vs. Predator - Graphics glitch

M-Walrus, 2021-12-24:
Round 7 has a graphics issue with a foreground door frame (about 2 minutes into the stage), as seen timestamped here:

https://youtu.be/Oo8bmBKDlFg?t=2174

Colors on the foreground section of door become two-tone and the priority shifts so it becomes a background object, making it look like players are walking on top of it.

Version tested is Euro region release, 10-24-21 core build on MiSTer FPGA

atrac17, 2022-03-20:
This may be due to overdump of GFX roms @jotego. I'll upload new MRA's here to test with the latest rom dump from mame.

jotego, 2022-04-20:
As this is a problem with the ROM dumps and not with the core, I will close it here.

closed 2022-04-20 by jotego

M-Walrus, 2022-05-11:
> As this is a problem with the ROM dumps and not with the core, I will close it here.

This was never confirmed @jotego. Do you have any further information about these bad rom dumps?  I am pulling from the most recent MAME set, verified.  Folks with ROM dumping capability should know if there is possibly no good dump of this game, if this is indeed the issue.

All we have in this issue report is a random person saying it *may* be due to some issue and they would test something, but no follow-up.

atrac17, 2022-05-11:
You've used this specific ROM? Made an MRA to load this over dumped GFX bank?

[cps2.cpp: corrected some overdumps for avsp - Guru](https://github.com/mamedev/mame/commit/163659b50aabc57aaeeb55e35e88d85523cdbc4c) 

When @theypsilon has the new distribution for ROM fetching available, I'll replace the MRA.

I'm not some random person. Since you did the above, you know that when you looked at the MRA, I'm one of the authors. This was discussed off of GitHub because it effects every single MiSTerFPGA user.

atrac17, 2022-05-12:
Core and hardware. Thanks tester nobody!

[image]

[image]

[image]

https://youtu.be/oA_aBrCciFI?t=2520