jotego/jtcps #117: CPS2: Sprite gaps (Multiple games)

lsnderick, 2021-12-25:
Recently I noticed there seems to visible "gaps" between some elements of the larger sprites. During the screen transitions those sprites seem to move out of sync with the rest of the image. So far I've seen this is in Vampire Savior 2 The Lord of Vampire (Intro), Vampire Hunter 2 Darkstalkers Revenge (Intro), Marvel Super Heroes (One of the stage background layers). Some screenshots:

[image]
[image]
[image]
[image]
[image]
[image]

jotego, 2021-12-25:
I think this is new. Maybe I broke something in an update. I don't remember seeing this problem before.

jotego, 2021-12-25:
The problem was introduced in commit 0fdcb02

jotego, 2021-12-25:
Fixed in c67b7e7

closed 2021-12-25 by jotego