jotego/jtcps #123: SF2: sprites drawn outside border on right hand side

fluxcorenz, 2022-02-05:
[image]

Foreground sprites are being drawn past the right hand side 'border', as described as the vertical line where the background is no longer drawn. This includes player characters, fireballs, crates etc.

A theory has been proposed that this is new behaviour since the shadowmask video processing has been added, but I have not confirmed this.

jtcps1_20220201

jotego, 2022-02-05:
I think you're right, at some point in the development the two layer lengths got mismatched. This is from Ghouls'n Ghosts:

[image]

core version f83bc43 (you can see this code in the about screen)

jotego, 2022-03-21:
Fixed

closed 2022-03-21 by jotego