jotego/jtcps #126: MiSTer CPS-2 Street Fighter Alpha/Zero 3 - Sound Test Graphical Glitch

MJY71, 2022-02-23:
In the 'Sound & Voice Test' screen for Street Fighter Alpha 3 and Zero 3, the column labels for channels F-8/8-F flash rapidly when not in use now.

jotego, 2022-06-25:
As far as I can tell, this must be the correct behaviour.

MJY71, 2022-06-25:
Interesting. I've used this to listen to the music for many years, and I have only seen this occur with more recent builds of the CPS-2 core over the last few months. Earlier builds do not exhibit this behavior. For what it's worth, MAME does not either, and never has to the best of my knowledge.

jotego, 2022-07-02:
Tested e289fc5. On sfa3 and sfaeud (Phoenix version)

For each song selected, the letters on top of silient bars flash between blue and yellow colors:
[image]

and

[image]

In MAME, the colours don't change. All the image in drawn in the SCR1 layer.

jotego, 2022-07-03:
Still occurs in bf5a34e after fixing the CPU decoding

MJY71, 2022-07-04:
I wish I could help narrow it down for you, but I'm now having trouble remembering exactly when I first noticed this behavior with the CPS-2 core. I am still convinced that it was only a few (3-4) months ago, possibly as far back to the beginning of this year, but I tried several releases going all the way back to a year ago, and that behavior is present for those, too... which makes me think perhaps something else is possibly causing it (MiSTer main?). This is a menu screen I have accessed quite frequently ever since the game was first playable in the core, and I have trouble believing that I would not notice something that obvious for that many months.

jotego, 2022-07-04:
Old commits (from older to newer)
- 2e38cbb won't boot (EEPROM ERROR)
- 3fce7fe bugged
- 48a929b bugged
- fb5c650 bugged
- 7f093bb bugged
I think this bug has always been there.

closed 2022-07-04 by jotego