jotego/jtcps #131: cps1 - 3 Wonders - intro glitch

barbudreadmon, 2022-05-24:
Seen on a friend's mister last week, i don't own one.

See pcb video at https://www.youtube.com/watch?v=UiJeLfruoh4&t=71s , the gun isn't supposed to be cut but it's cut on mister.

atrac17, 2022-06-19:
Reference #132 

Last working build is [20211206](https://github.com/jotego/jtbin/blob/master/mister/cps1/releases/jtcps1_20211206.rbf) (JTCPS1 v9f4b585).

[image]

Current build is [20220601](https://github.com/jotego/jtbin/blob/master/mister/cps1/releases/jtcps1_20220601.rbf) (JTCPS1 v40add73).

[image]

jotego, 2022-06-25:
Thanks @barbudreadmon for the PCB video.
Thanks @atrac17 for providing the history information. I've reverted the compromising change in edd0cb9. This fixes this issue but reopens #40.

commit edd0cb92 (2022-06-25): Revers partly 0fdcb02.
Reopens issue #40
Closes #131 and #132
closed 2022-06-25 by jotego