jotego/jtcps #132: Final Fight: check layer alignment

jotego, 2022-06-04:
Check whether this is correct

[image]

atrac17, 2022-06-19:
The priority issues are present in other CPS1 titles. Final Fight is not the only title effected. I was able to trace this back to the last known working build.

Last working build is [20211206](https://github.com/jotego/jtbin/blob/master/mister/cps1/releases/jtcps1_20211206.rbf) (JTCPS1 v9f4b585).

[image]

[image]

Current build is [20220601](https://github.com/jotego/jtbin/blob/master/mister/cps1/releases/jtcps1_20220601.rbf) (JTCPS1 v40add73).

[image]

[image]

Traced back to the following commits:

[4cef041](https://github.com/jotego/jtcps/commit/4cef041467b0fea679848870385c073a28a18d8f)
[0fdcb02](https://github.com/jotego/jtcps/commit/0fdcb021b95a30dc7bf548203660dcb2f20e56e7)

jotego, 2022-06-25:
Thanks @atrac17 for providing the history information. I've reverted the compromising change in edd0cb9. This fixes this issue but reopens #40.

closed 2022-06-25 by jotego