jotego/jtcps #133: Night Warriors: raster effect?

jotego, 2022-06-05:
From Professor Jones in Discord

> I found another small graphical bug, in Night Warriors (World), using the latest CPS2 core (20220601) : in Pyron's stage, when descending from a jump, there's an horizontal line that forms for a brief time (seems like the layer is "separating"). I'm sorry for the bad quality of the pictures, I had to take screens from a video. Here's a link to the [video](https://photos.app.goo.gl/enZbxzoC8Ziv2izz6).

[image]

atrac17, 2022-06-19:
Please see [here](https://github.com/JTFPGA/jtalpha/issues/28). This was verified as being a function on real hardware back in July of 2021. Reference [video](https://www.youtube.com/watch?v=dqUuGBbKHaU) is here.

jotego, 2022-06-25:
Thanks for the references. I'll close the issue.

closed 2022-06-25 by jotego