jotego/jtcps #134: Street Fighter II' Chamion Edition alternatives: Multiple bootleg games have issues

AMurderOfCrows, 2022-06-06:
Street Fighter 2 Mix (Ver 1.2 220126) <-- missing a bin: games/hbmame/sf2mix.zip/smx-13m.6c not found
Street Fighter II' Champion Edition -Accelerator!, bootleg, set 1- <-- missing a bin: games/mame/sf2acc.zip/s92_13.bin not found
Street Fighter II' Champion Edition -Accelerator!, bootleg, set 2- <-- missing a bin: games/mame/sf2acca.zip/s92_13.bin not found
Street Fighter II' Champion Edition -bootleg with level selection- <-- boots to black screen
Street Fighter II' Champion Edition -Dstreet-6, bootleg- <-- missing a bin 6st.u29 not found
Street Fighter II' Champion Edition -M10, bootleg- <-- boots to black screen
Street Fighter II' Champion Edition -M3, bootleg- <-- missing a bin: games/mame/sf2m3.zip/s92_13.bin not found
Street Fighter II' Champion Edition -Mstreet-6, bootleg, set 1- <-- missing a bin: games/mame/sf2cems6a.zip/ms6.u29 not found
Street Fighter II' Champion Edition -Mstreet-6, bootleg, set 2- <-- boots to black screen
Street Fighter II' Champion Edition -Mstreet-6, bootleg, set 3- <-- boots to black screen
Street Fighter II' Champion Edition -Rainbow, bootleg, set 3- <-- missing a bin: games/mame/sf2rb3.zip/s92_13.bin not found
Street Fighter II' Champion Edition -RE, bootleg- <-- boots to black screen
Street Fighter II' Champion Edition -Red Wave PtII, bootleg- <-- missing a bin: games/mame/sf2redp2.zip/s92_13.bin not found
Street Fighter II' Champion Edition -Red Wave, bootleg-  <-- missing a bin: games/mame/sf2red.zip/s92_13.bin not found
Street Fighter II' Champion Edition -Taiwan 920313- <-- missing a bin: games/mame/sf2cet.zip/s92-13m.6c not found
Street Fighter II' Champion Edition -UPL bootleg-  <-- boots but no input controls and missing graphics or graphic layers
Street Fighter II' Champion Edition -V004 <-- missing a bin: games/mame/sf2v004.zip/s92_13.bin not found

This isn't the end of the list, but with this many problems i figure that maybe all the alternatives are worth getting reviewed.

the MiSTer discord members helped me get some troubleshooting done:

Zakk: I bet most of these jsut have the wrong parent set in the mra
Zakk: sf2 accelerator bootlegs have the parent set as 'sf2.zip' but it should be 'sf2ce.zip'
Zakk: you can probably edit the mras, yes. for instance in those accelerator sf2 ones you can change the 'sf2.zip' to 'sf2ce.zip' and it will likely work

atrac17, 2022-06-19:
For the "missing a bin" section, this is a ROM issue or the xml (ROM) has changed since the MRA was created. Some sets have protection issues and won't work on real hardware (hence a black screen), others use kludge code for GFX. Some are modified hardware based and "bootlegs". There are a few that probably need to be verified against the XML.

[Here](https://github.com/atrac17/Arcade_Offset/tree/main/release/_Arcade%20Offset/_CP%20System) are a few bootlegs that aren't available. That UPL bootleg will never work. Rainbow Edition set 1 and 2 require patches to run on real hardware; set 1 is included in this repo. The M bootlegs require additional GFX code not found on original hardware.

This isn't prioritized like mame amoc, some of them won't work period without heavily modifying the core. I finally got koryu's layer priorities working with kludge 1. I'll scrub whatever is in SF2, SF2CE, SF2HF.

jotego, 2022-07-02:
What @atrac17 commented is accurate. Some bootlegs run on different hardware and I am not giving support to them now.

closed 2022-07-02 by jotego