jotego/jtcps #137: CPS-2 graphical corruption on left side of screen

MJY71, 2022-06-28:
The most recent CPS-2 update (jtcps2_20220627	08141df) exhibits graphical corruption in the left-most column of pixels on several (all?) games. Some layers do not extend all the way to the left edge of the screen, and there is occasional color shifting with some layers in that first column. It is very noticeable during the Street Fighter Alpha/Zero 3 intro sequence.

jotego, 2022-07-02:
From sfa3:

[image]
The object layer seems to start off a bit too early (at least 2 pixels)

In the Ryu image, there is a gap because the object layer (upper face) is off to the left by at least 1 pixel. The lower face is SCR2

[image]

From sfa2
Chun Li's is drawn in the OBJ layer and starts a bit too early
[image]

jotego, 2022-07-02:
The problem came because the cawing fix had moved the SCR layers ahead one pixel, and the CPS2 colmix was modified to add a pixel delay to the scroll (see #117). After reverting the cawing fix edd0cb9, the SCR layer had to be taken without delay in CPS2 colmix. e289fc5 fixes this.

commit e289fc5e (2022-07-02): Fixes #137 and #117 (again) after revertint partly 0fdcb02
closed 2022-07-02 by jotego