jotego/jtcps #141: wrong sound in king of dragon and captain commando

titorin0, 2022-09-14:
In the first stage of captain commando the cymbals are perceived very strident.
they are not corrected by touching the filters.

the same goes for the king of dragon in stage 4 .
becoming unpleasant
[https://youtube.com/clip/UgkxhsKSqb7QJffJFQkWv4YaObqXCOb_Gsxo](url)
[https://youtube.com/clip/UgkxO7wt1Am9h_BWJhiZ_K9NUSGeHzs6rsx1](url)

nota: perdón si he cometido alguna falta en inglés 
note: sorry if I have made any mistakes in English

TesterNes, 2022-11-12:
That Captain Commando's video is not from a real pcb but MAME32Plus!Plus!0.116u3
This is from a real one: https://www.youtube.com/watch?v=0o9Utr5rjFQ
Anyway, this is related to https://github.com/jotego/jtcps/issues/145

jotego, 2022-11-21:
Close as duplicated. We will track it in #145

closed 2022-11-21 by jotego