jotego/jtcps #144: Puzzloop2/CPS2 Spinner support

eedomus, 2022-11-06:
By Default, Puzzloop is configured to be played with a joystick.

But you can enable spinner control going to "Test Menu / Configuration / Operation"

Playing this game with a spinner is a nice added value on the gameplay 🥰

[image]

[image]

jotego, 2023-01-15:
The spinner logic is currently not connected to the CPU. Apparently, there are two games using the spinner and the hardware is slightly different.

To do:

- [ ] connect spinner to JTCPS2
- [ ] Eco Fighters mode
- [ ] Puzz Loop 2 mode

capitaineflam25, 2023-01-16:
Thanks for the explanation 👍🏻

jotego, 2023-01-18:
Completed on e1c67c9
To be released this Friday

closed 2023-01-18 by jotego