jotego/jtcps #146: Speed increase with SF2 Hyper Fighting

sqwirral, 2022-11-22:
The speed of SF2HF has increase at some point. It was good back at 20210131.

Here is a video (recorded by Ohmi) comparing the current 20220819 to a real HF board:
https://drive.google.com/file/d/1FNYcXS0yum-oKHpksb2RO4Xx8hH1cAg5/view?usp=share_link

I tested and found the speed is faster between 20220819 and the previous 20220705 release. Here is my video of those 2 side by side:
https://discord.com/channels/558531469128237076/716238276679303208/1043657774992719982

Both 20220819 and 20220705 tested faster than the old 20210131 version. Here's my screenshot when I synced videos of all 3:
[image]
(20220819 on 62 seconds, 20220705 on 65, and 20210131 is only on 67)

jotego, 2022-12-16:
Some changes in how the /DTACK signal is handled have affected the core performance over time. This is a collection of the different values recorded using _SF2 DMA no b3.mra_ These tests measure the CPU speed under different DMA settings. In this table I am just comparing the default value. Note that the numbers are in hexadecimal.

Version         | commit  | Speed     | JTFRAME
----------------|---------|-----------|-------------------------
Original PCB    |  N/A    |  2C00     |
20210131        |         |  2C04     |
20210604        |         |  2C0A     |
20210828        |         |  2C0B     |
20210921        |         |**357C**   |
20211002        |         |  357C     |
20211024        |         |  357C     |
20220112        |         |  3584     |
20220321        |         |  3584     |
20220417        |         |**2BF9**   |
20220422        |         |  2BF9     |
20220427        |         |  2BF9     |
20220601        |         |  2BF9     |
20220627        |         |  2BF9     |
20220704        |         |  2BF9     |
20220705        | 61626fc |  2BF9     |  bd7cf34
20220819        | b78bdae |**3576**   |  a108a4a

The immediate action is to [delete version 20220819](https://github.com/jotego/jtbin/commit/7059c052a419959bf5c3fdd335b4b7bfc7c10126). I keep the issue open until the source code incorporates the fix.

jotego, 2022-12-22:
- The changes in the jtframe_68kdtack module in [JTFRAME](https://github.com/jotego/jtframe/commit/fa2d6a1a78d3d41b57ae12030c54f8096d01d8b0)  caused the speed increase
- Fixed in bd63249

commit bd632498 (2022-12-22): Fix #146
closed 2022-12-22 by jotego